Ultimaker/Cura #13032: Agregar Funciones.

JorgeNadeR1, 2022-08-14:
### Is your feature request related to a problem?

Al hacer zoom siempre se aumenta o reduce en un mismo punto de coordenada.

### Describe the solution you'd like

Poner la función de zoom en el cursor para que al ampliar o reducir se realice donde este el mismo. 

### Describe alternatives you've considered

Activar o desactivar está opción.

### Affected users and/or printers

Se beneficiaran todos los usuarios.

GregValiant, 2022-08-15:
In Cura, go to "Preferences / Configure Cura" and in "Viewport Behavior" are "Zoom towards mouse direction".  Is that enabled?
("Zoom toward mouse direction" only works when you are in "Perspective" view.  It does not work when in "Orthographic" view.)

MariMakes, 2022-09-22:
Hey @JorgeNadeR1,

I think @GregValiant is right! We have a Zoom Towards Mouse Direction in Cura. 
In the Spanish version of Cura it's called: Hacer zoom en la dirección del ratón

Can you please confirm if this is the feature you are looking for? 
Then we can close the ticket. 😄 

Gracias 👍

closed 2022-10-06 by github-actions[bot]