Ultimaker/Cura #13033: Tree Supports not generating for overhang

jpirok, 2022-08-14:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

When slicing the attached stl with tree supports and the large porting of the object touching the build plate, there is an overhang that does not get a support generated for it during slicing. Slicer is set to generate tree supports touching the build plate. The project file with settings is attached.

### Actual results

[image]
[image]

Notice the unsupported overhang at the top in the second picture.

### Expected results

The overhang at the top of the of the object should have a support generated by the slicer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Frame.zip](https://github.com/Ultimaker/Cura/files/9334054/Frame.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9334059/cura.log)

GregValiant, 2022-08-14:
Thanks for the report.
There are over 70 settings for supports with 5 settings specific to tree supports.
Changing the "Tree Support Branch Angle" to 25° makes a big difference but it isn't the only change you need to make.

I don't see a bug here.  You need to make changes to some of your settings to get tree supports to work with this model.
[image]

For that model I think Normal Supports do a better job.  If you make the floor distance 2X layer height the bottom will come off pretty easy.  Using tree supports isn't always the best way to print a model.
[image]

I'll leave this open so you can comment and remove the Bug label as it appears to be a settings issue.

jpirok, 2022-08-19:
Got it. Thank you!

closed 2022-08-19 by jpirok