Ultimaker/Cura #13034: Build from source Windows 10 SCM getting sources from url auto

phedg1, 2022-08-14:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker Doesnt Matter

### Reproduction steps

I am on the latest commit for the 5.1.0 branch for both curaengine and cura.
I created a workspace directory, with the Cura and CuraEngine folders in its root.
I had to manually create a folder called "venv" in the root of the cura directory.

Following the build from source page on the wiki.
I installed the dependencies.
Then I exported the curaengine package.
Then I exported the cura package.
Then I tried to install the cura package with the curaengine dependency override.

This command has to be changed from the build from source wiki page because --requirement-override doesn't exist, but --require-override does.

`pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect
conan remote remove cura

conan export-pkg . cura/5.1.0@peter/testing --force
conan create . curaengine/5.1.0@peter/testing --build=missing --update
conan install cura/5.1.0@peter/testing --build=missing --update --require-override="curaengine/5.1.0@peter/testing" -g VirtualPythonEnv --json cura_inst/conan_install_info.json -if cura_inst`

### Actual results

Exporting both the curaengine and cura packages worked fine.
Installing the cura package with the curaengine override returns the following error:

`arcus/5.1.0: Already installed!
pynest2d/5.1.0: Already installed!
savitar/5.1.0: Already installed!
curaengine/5.1.0@peter/testing: Already installed!
uranium/5.1.0-beta+testing_75@ultimaker/testing: Already installed!
cura/5.1.0@peter/testing: WARN: Build folder is dirty, removing it: C:\Users\Peter\.conan\data\cura\5.1.0\peter\testing\build\6b2ef4c2a9c58a78e82a84d87fe518793feb094d
cura/5.1.0@peter/testing: WARN: Trying to remove corrupted source folder
cura/5.1.0@peter/testing: WARN: This can take a while for big packages
cura/5.1.0@peter/testing: Configuring sources in C:\Users\Peter\.conan\data\cura\5.1.0\peter\testing\source\.
cura/5.1.0@peter/testing: SCM: Getting sources from url: 'auto'

JSON file created at 'C:\Users\Peter\Documents\3D_Print\_Software\Workspace\Cura\cura_inst/conan_install_info.json'
ERROR: Couldn't checkout SCM: Command 'git -c http.sslVerify=true clone "auto" .   ' returned non-zero exit status 128.
fatal: repository 'auto' does not exist`

### Expected results

I expected this command to just run perfectly fine, but the command does not complete.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Running this command without the --require-override will finish without errors, but I need both a custom cura and custom curaengine for the changes I'm envisioning to the slicing.

phedg1, 2022-08-14:
Strangely though, if you edit conandata.yml to use "curaengine/5.1.0@peter/testing" instead of "curaengine/5.1.0", then remove the --require-override argument it works just fine.

MariMakes, 2022-08-26:
Hey @phedg1 ,

Welcome to the elite Running From Source Club! 👑 
I'm not very knowledgeable about how to run from source. 

But I do know that we updated our documentation a few days ago. 
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

Can you see if this has the solution you are looking for? 
And if not can you leave a comment and tag @jellespijker ? 

Thank you!

closed 2022-09-09 by github-actions[bot]