Ultimaker/Cura #13035: Different infill density for first layer above infill

DuckersMcQuack, 2022-08-15:
### Is your feature request related to a problem?

I want to save plastic, to have say 10% infill for the entire model, except in the area that will have a first layer of a different part. The bottom of a cup for example. It got really rough for the first layer due to the low percent infill.

[image]

### Describe the solution you'd like

Have 2 infill options. One for general model, one for a part that will be the floor of a cup for example. 

### Describe alternatives you've considered

Higher density overall infill, but that uses more plastic.

### Affected users and/or printers

Everyone.

GregValiant, 2022-08-15:
Have you looked at "Gradual Infill Steps"?  It will start out at low density and build up as it gets near the roof so the roof is well supported.  "Gradual Infill Step Height" becomes available when you have steps >0.

DuckersMcQuack, 2022-08-15:
> Have you looked at "Gradual Infill Steps"? It will start out at low density and build up as it gets near the roof so the roof is well supported. "Gradual Infill Step Height" becomes available when you have steps >0.

Guessing that's only a cura 5 feature? Cause i'm waiting for 5 to be bug fixed a bit before i installed it as my test prints with it with my good 4.12 when it released delivered quite some bad results.

Asterchades, 2022-08-15:
It's been in place since Cura 4 at least, but you'll either need to search for it or adjust setting visibility to access it. Note that it's not an on-or-off setting, rather it defaults to zero to indicate it is not in use.

It functions by halving the Infill Density at the specified intervals, the specified number of times. So if you want a higher density at the top you'll need to set your overall density higher accordingly. eg Setting the Infill Density to 50% with two Gradual Infill Steps and a Gradual Infill Step Height of 1mm would result in the first millimetre below the skin being at 50%, the next millimetre at 25%, and the remaining infill at 12.5%.

MariMakes, 2022-09-22:
Hey @DuckersMcQuack,

There are indeed three features that mimic this behavior as well 😄 

We have Gradual Infill that has been in Cura for at least 4 years.
[image]

We have Lightning Infill pattern, introduced in Cura 4.12
[image]

And we have Skin Edge Support layers. I think those will help you most in this case
[image]

Can you let us know how your request is different from what we already offer?
Else I will close this issue. Thank you 😄

closed 2022-10-08 by github-actions[bot]