Ultimaker/Cura #13038: Cura not slicing a file, but slices other files from the model with no issues.

Mikyleee, 2022-08-15:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Artillery Sidewinder X1 V4

### Reproduction steps

Exported STL from Solidworks (tried very fine detail, fine, coarse and custom).

Imported STL to Cura.

Hit Slice. 

### Actual results

I get this dialogue. 
[image]

I have various other solid bodies in this part file that when I export to STL, they work ok. It's just this individual part.

I also reopened the STL and part file in SW and ran the geometry analysis and no issues. 

### Expected results

Slice correctly?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Chess Board Insert.zip](https://github.com/Ultimaker/Cura/files/9336599/Chess.Board.Insert.zip)

GregValiant, 2022-08-16:
Thanks for the report.
I checked the STL model in MS 3D Builder and it did not report ay errors.  I imported the STL into Cura and it sliced with a CR-10 as the active printer and my own profile.

Using your 3mf file, it would not slice.  Changing the "Ironing Pattern" from "Concentric" to "ZigZag" allowed the model to slice.
I think this report makes it unanimous...there are problems with the Concentric pattern: as Infill, as Top / Bottom, as Support, and now as Ironing.  I'll let the developers decide if this is actually a duplicate.  It does appear to be a continuing problem with the Concentric pattern though.

Mikyleee, 2022-08-20:
Hi Greg, sorry for the slow reply. 

Many thanks, that seems to have solved it... quite frustrating as the rest of it has already been printed with concentric ironing and concentric support interface too.

None the less, at least I can print it now! Thanks! :D