Ultimaker/Cura #13039: Error slicing model, no detailed explanation of what's wrong

stevenwbuehler, 2022-08-15:
### Application Version

5.1.0

### Platform

Windows 11, i7-1165G7

### Printer

Creality Ender 3

### Reproduction steps

Loaded model, attempted to slice. Applied 20% infill, gyroid infill, tree support from build platform

### Actual results

Received slicing error, no detail

### Expected results

Model should be sliced and prepped for printing.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Don't know how to obtain log file
[CE3_FreziGrant-185.zip](https://github.com/Ultimaker/Cura/files/9337308/CE3_FreziGrant-185.zip)
.

fieldOfView, 2022-08-15:
> Don't know how to obtain log file

See https://github.com/Ultimaker/Cura/#logging-issues

stevenwbuehler, 2022-08-15:
[cura.zip](https://github.com/Ultimaker/Cura/files/9337329/cura.zip)
Would help if I read things, wouldn't it? Latest log attached.

GregValiant, 2022-08-15:
Thanks for the report.
This is a duplicate of several other reports.  The model appears to be error free but won't slice in the current orientation on the build plate.
As a workaround, move the model on the build plate about 10mm in the X and about 10mm in the Y and rotate the model 90° or so.  I did get it to slice after doing that.
[image]

Vandresc, 2022-08-16:
We have a ticket on our backlog, devs CURA-9380 for our internal reference. 

Thank you for your feedback!

stevenwbuehler, 2022-08-16:
Sure enough, shifting and rotating the model as described above worked. She's just about done printing now 17h later. Thanks for the quick turn-around.

GregValiant, 2022-08-16:
@stevenwbuehler I'm going to leave this open in case the developers need another example file.
I'm glad the workaround fixed it, but we shouldn't need a workaround.