Ultimaker/Cura #13042: Transition layer between bottom and vase mode

catapulta, 2022-08-15:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I'm printing with a large nozzle and high layer height. Set a line width of 4.0 mm and 0.8 mm layer height.

### Actual results

I get an overextruded, rim-like, horizontal line upon vase mode transition. The issue is over extrusion when vase mode switches from "bottom layers" to spiralized vase.

Debugging gcode: I reviewed the gcode and in the same transition layer it does a regular skin print first (flat, no Z changes), and then prints a first layer of spiralized mode (progressive Z increases). Problem is the Z height does not change between the end of the bottom skin layer and the start of the spiralized vase! I'm guessing Cura does it to ensure good adhesion but with a large nozzle the results are bad.

Example GCODE of a sliced cube with 4 bottom layers of 8 mm each. Skin will be printed once flat, and once again using spiralize without raising the extruder:

```
;MESH:NONMESH
G0 F300 X99.506 Y135.494 Z4; starts at Z4
G0 F9000 X101.07 Y133.93
G0 X133.93 Y133.93
G1 F2700 E3167.96495
G0 F9000 X135.5 Y135.5
;TIME_ELAPSED:125.707592
;LAYER:4
;TYPE:WALL-OUTER
;MESH:Cube
G1 F1500 X99.5 Y135.5 E3215.85955
G1 X99.5 Y99.5 E3263.75414
G1 X135.5 Y99.5 E3311.64873
G1 X135.5 Y135.5 E3359.54333 ; ends at Z4
G1 X99.5 Y135.5 Z4.2 E3407.43866; starts full extrusion at Z4 (again!)
G1 X99.5 Y99.5 Z4.4 E3455.334
G1 X135.5 Y99.5 Z4.6 E3503.22933
G1 X135.5 Y135.5 Z4.8 E3551.12466
;TIME_ELAPSED:137.452840
;LAYER:5
```

This issue is related but not the same as #11590 since this is not just ramp up but duplication of volume.

### Expected results

An invisible transition from bottom layers to vase mode spiralized layers (without a horizontal over-extruded line).

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Attaching Cura project:
[cube.zip](https://github.com/Ultimaker/Cura/files/9338776/cube.zip)

GregValiant, 2022-08-15:
Thanks for the report.
That initial layer over-extrusion is a known issue.  With smaller nozzles it isn't noticeable but with large nozzles it can leave a ridge.
The @smartavionics fork of Cura has a compensation for that first layer of over-extrusion during spiralize.

catapulta, 2022-08-17:
Thanks for the pointer! Do you happen to know if there's any plans to fix/add an option in main Cura? Even if we could choose to disable the "first part" of the layer duplicating the volume it'd be a win for large nozzles

GregValiant, 2022-08-17:
I don't know what the Cura Team has planned as I'm just the greeter here.
You could work around this manually.  It would require knowing about where in that first spiralize layer the relative 12:00, 3:00, 6:00 and 9:00 positions of the nozzle are.  By entering M221 lines at appropriate points you could step the flow up until at about the 9:00 position you are back to M221 S100.
Within a text editor - If you look at the line numbers of that layer and it takes 10,000 lines then you could split that into 1/4's and enter the M221 lines at appropriate intervals.  It would take a bit of calibrating but I think it would be doable.

I have used the SmartAvionics fork.  It's based on Cura 4.x and that Spiralize fix seems to work well but I can't really tell as I generally print with smallish nozzles.

catapulta, 2022-08-22:
Hi @GregValiant, thanks for the follow-up -- I like the idea of the clockwise flow control, but I'm having a hard time finding those positions. Perhaps it'd be easier to parse the whole layer to find the total length and split it by line with M221?

Also, I just tried @smartavionics' fork, it does compensate nicely in the first spiralized layer. I'll be using this branch going forward. Amazing work!

Posting this break-down table of the transition layer based on the example above in case anyone finds it useful:

<html>
<body>
<google-sheets-html-origin>

GCODE | X | Y | Z | E | E delta
-- | -- | -- | -- | -- | --
Prev. Layer End |   |   | Z3.2 | E2470.43411 |  
G1 | X98.7 | Y86.2 | Z4 | E2500.44806 | 30.01395
G1 | X136.3 | Y86.2 | Z4 | E2530.462 | 30.01394
G1 | X136.3 | Y123.8 | Z4 | E2560.47595 | 30.01395
G1 | X98.7 | Y123.8 | Z4 | E2590.48989 | 30.01394
2nd part of same layer
G1 | X98.7 | Y86.2 | Z4.2 | E2603.62118 | 13.13129
G1 | X136.3 | Y86.2 | Z4.4 | E2622.38016 | 18.75898
G1 | X136.3 | Y123.8 | Z4.6 | E2646.76684 | 24.38668
G1 | X98.7 | Y123.8 | Z4.8 | E2676.78121 | 30.01437

</body>
</html>