Ultimaker/Cura #13044: Cura outputting unopened files in gcode

jwolfm98, 2022-08-15:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anycubic Mega X

### Reproduction steps

Open Cura, open file, select an STL file, multiply selected by some number 2-6, slice, save the gcode file anywhere, close Cura. Repeat.

### Actual results

There is an inconsistent tendency for Cura to output a gcode file that actually has mesh references to files that were previously opened on the previous slice.

This was discovered when mid-print the printer moved to a new location and started printing an entirely different object.

In the gcode file the file name for mesh: may not appear as it should. In slices that have duplicate objects, you might see "filename (1)", which is normal. But sometimes I'll also find "totally different filename" halfway down in the gcode.

SD cards were tested, formattered, error checked. Save to Disk produced the same bug. Closing and re-opening Cura sometimes fixes it, sometimes Ive had to do that multiple times.

### Expected results

I assumed that if I open Cura, open a file, use multiply selected and save the file. The print will only print the object file I opened, and only the number of objects that have been put into the print area.

If I close Cura and repeat these steps for a new filt, I shouldt see a filename of an object that I have not opened in that instance of Cura.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Specific project files do not manipulate the results. Certain settings don't seem to impact results. Freshly rebooting Windows can still generate the problem on the first time opening Cura.

I can upload files later this evening

GregValiant, 2022-08-15:
Thanks for the report.
Please include a project file, the Cura.log file, and a gcode file (or two) from the SD card.  You seem like you are aware that the symptoms you have are often SD problems.  I'm not saying it is here.  I'm just saying.

jwolfm98, 2022-08-16:
I will get the logs as soon as I can re-create it.... It's happened enough that I know it can be repeated, but actually reproducing it is still a bit of a mystery. In one instance, I saved both to an SD Card and to my local drive - both had the same problem. I'll update as soon as I see it repeated again.