Ultimaker/Cura #13049: (5.1) Crashes when loading files

sebastlamon16, 2022-08-16:
### Application Version

5.1

### Platform

Windows 10

### Printer

Creality CR-10s

### Reproduction steps

1. Tried to open any slt or gcode files using various methods (drag, right click/open with, open file from within cura)
2. Tried restarting computer
3. Tried reinstalling cura

### Actual results

Starts to load the files then promptly crashes

### Expected results

Showing me a pretty little 3d model in the middle when loading a file would be a good start and "not crashing" would also be in expected results range.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

This is following a update from 4.7 to 5.1 which was also it's own little pain to deal with. In the end uninstalling every versions then deleting anything remotely affiliated with cura and finally reinstalling it did the trick.

I do not have either chitubox or octoprint on my PC which looking online seems to be the only cause for my problem people have found
[cura.log](https://github.com/Ultimaker/Cura/files/9354604/cura.log)
.

Vandresc, 2022-08-19:
Hi @sebastlamon16, could you also share a project file? Logs don't really show anything weird explaining your issue. Thank you!

sebastlamon16, 2022-08-19:
Well here's a project file of nothing, since loading anything else makes it crashe it's all i can give really

[project file.zip](https://github.com/Ultimaker/Cura/files/9382724/project.file.zip)

sebastlamon16, 2022-08-19:
Ok, im just not 100% sure which file exactly you want is it this one?

[cura.zip](https://github.com/Ultimaker/Cura/files/9384276/cura.zip)

Vandresc, 2022-08-22:
Ah sorry, I was thinking more of the 5.1 folder found in C:\Users\name\AppData\Roaming\cura, where you store all your user changes, machine definitions, custom profiles, and everything related to top user changes. I hope that replacing my config with yours will bring Cura to a state where I can reproduce the issue.

sebastlamon16, 2022-08-23:
Aaah I see, there you go then.
[5.1.zip](https://github.com/Ultimaker/Cura/files/9399469/5.1.zip)

MariMakes, 2022-08-30:
Hey @sebastlamon16,

I took a look at your configuration folder and your logs and I don't see anything odd.
When I load your configuration and a model I don't get a crash. 😞 
[image]

It's a shot in the dark, but could it be related to a Nahimic Audio Driver as described here?
https://www.youtube.com/watch?v=XW-E1P6Ms4s

sebastlamon16, 2022-08-31:
I looked at the video but sadly it doesn't seems like i have Nahimic audio on my pc, nor do i use a second monitor for the that matter. I think i will just have to revert to Cura 4 for the time being. Thanks for the help.

MariMakes, 2022-08-31:
Thanks for the update. 
I wish we could help you better 😞 
But I'm glad that you can still use Cura 4.x

closed 2022-08-31 by MariMakes