Ultimaker/Cura #13050: Line/ZigZag bottom in spiral mode

mbooyzen, 2022-08-17:
### Application Version

5.1.0/5.0.0

### Platform

Windows11, Also saw it on win10

### Printer

Ender3, Applicable to more than one 

### Reproduction steps

Selecting spiralize outer contour
selecting bottom layer pattern

### Actual results

Bottom pattern does not change when selecting lines as bottom layer pattern, stays concentric. 

### Expected results

Needs to change to lines/zigzag/concentric as is selected

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-08-17:
Thanks for the report.
Yes, the new default is "Concentric" and that is all that is available for the bottom pattern when using Spiralize.
This has been noted before and people aren't happy.  I'll mark it as a duplicate and leave it as a Bug report although at this point it may also qualify as a New Feature request since it is behaving as designed.

mbooyzen, 2022-08-17:
Thanks, this feature was in 4.x and helps getting a water tight bottom. Concentric is also not as strong.