Ultimaker/Cura #13051: slicing issure

Christoph8300, 2022-08-17:
### Application Version

5.1.0

### Platform

Windows 11 22000.856

### Printer

Creality Ender3 V2

### Reproduction steps

I tried to slice an big and difficult object.
(sorry for my bad English if i wrote something wrong)

### Actual results

The Slicing Failed.
[image]

### Expected results

The Slicing not fails.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Dragon.zip](https://github.com/Ultimaker/Cura/files/9358860/CE3_Dragon.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9358870/cura.log)

GregValiant, 2022-08-17:
Thanks for the report.
It sliced fine for me and so I am unable to duplicate the problem.  Someone from the Cura Team will check the log file to determine what happened when you tried to slice it.

Christoph8300, 2022-08-17:
> Thanks for the report. It sliced fine for me and so I am unable to duplicate the problem. Someone from the Cura Team will check the log file to determine what happened when you tried to slice it. (BTW I thought it was going to the the "Crystal Dragon" which has a LOT off errors in it. This model has no errors as far as I could determine with MS 3D Builder.)

Thank you for the Answer. The weird thing is i have allready printed the Dragon 2 Times with no errors haha.

GregValiant, 2022-08-17:
You can double check your settings (make sure they are "All" visible) to make sure there are no red boxes indicating illegal entries.  That's about it.  Maybe the log file will show something.
There are been several reports about models with no errors that don't want to slice.  Moving the model on the build plate or rotating the model can get them to slice.  The Cura Team will be looking into that particular problem.