Ultimaker/Cura #13052: Incorrect slicing in 5.1.0

etherbrat, 2022-08-17:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Wanhao i3 Mini

### Reproduction steps

1. Created a composite of two stl files and saved as 3mf project
2. Sliced using fine settings
3. Saved gcode for printing

### Actual results

Cura 5.1.0 says the stl is watertight but does not produce a watertight slice of the 3mf. The uppermost layer is missing as can be seen in the gcode file. Cura 4.6.1 produces the correct result.

### Expected results

Watertight print.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file:
[0B-TwoSectors.zip](https://github.com/Ultimaker/Cura/files/9358872/0B-TwoSectors.zip
Incorrect output from 5.1.0: 
[image]
Correct output from 4.6.1: 
[image]

GregValiant, 2022-08-17:
Thanks for the report.
Your Initial Layer Height is 0.3 and your Layer Height is 0.1 so in the gcode "LAYER:0" goes down at 0.3 and "LAYER:1" is going down at a height of .4 and so the total number of layers appears to be off as there is no layer at 0.1 or at 0.2.  The Z height is correct.
This is the start of your final layer from the Gcode file.
G0 F600 X87.681 Y57.293 **Z8**
-------travel moves------
;TIME_ELAPSED:8384.368901
;**LAYER:78**
As a check...If you change your Initial Layer Height to .1 you will see that the layer numbers are what you expected to see.

I don't see a bug here as what is going on appears to be a function of your settings.  I'll remove the bug label but leave this open.  Let me know if that is OK.

etherbrat, 2022-08-17:
Many thanks for your quick response. Rank amateur here, so I’ll follow your advice and apply your principles in future. Cheers, NH

Neil Higgins (iPhone)
***@***.***

GregValiant, 2022-08-18:
We all start somewhere.  I'll close this for you as it appears to be figured out.

closed 2022-08-18 by GregValiant