Ultimaker/Cura #13054: Spiralize outer contour does not appear to use wall speed correctly

pufferfish-tech, 2022-08-17:
### Application Version

5.1.0-beta+304

### Platform

Windows 10

### Printer

FLSUN V400

### Reproduction steps

If you crank the speed up to 300 (yes, this is a real speed! :) ) then the wall speed still seems to be around 40-50. I'm not sure where this is coming from since I've increased all speeds to test. 

### Actual results

You get a much lower speed. It seems like maybe there is a speed limit around 50-60mm/s
[FV_001_60mm_solid.zip](https://github.com/Ultimaker/Cura/files/9359627/FV_001_60mm_solid.zip)
 

### Expected results

I would expect the vase mode wall speed to be = the outer wall speed. 

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

see attached images. 
With spiralize off, the wall speed is correctly set: 
[image]

Can also reproduce on 5.0.0

smartavionics, 2022-08-17:
Perhaps you have too large a value for Minum Layer Time?

pufferfish-tech, 2022-08-17:
> Perhaps you have too large a value for Minimum Layer Time?

spot on! Thank you! 
[image]

closed 2022-08-17 by pufferfish-tech