Ultimaker/Cura #13058: dual extruder bot extruders heat up no common heater check box to uncheck

itssllc, 2022-08-17:
### Application Version

5.1

### Platform

windows10

### Printer

gmax 1.5xt

### Reproduction steps

all I could do was search for settings to see if any were checked for left extruder. There were none,  in maage printers I think there used to be a check box for dual extruders that said common heater block...not sure if that is it or not.

### Actual results

have to manually lower temperature on extruder 1, not using it.

### Expected results

to run with only extruder 2

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

recently installed version 5.1
[G15PDE_raised star figit KR.zip](https://github.com/Ultimaker/Cura/files/9366041/G15PDE_raised.star.figit.KR.zip)

itssllc, 2022-08-18:
> Thanks for the report.
> The setting was moved a few versions ago.
> Go to the Cura MarketPlace and find and install the plugin "Printer Settings". When you restart Cura you will see a new heading has been added below "Experimental".
> Within the Printer Settings are "Extruders share heater" and "Extruders share nozzle". They are usually used together.
>  <https://user-images.githubusercontent.com/64202104/185264809-20bdf8cb-3ccc-4aa9-b270-27c2295b0714.png>

Thank you for the response, that did show the setting like you said but it did not correct the trouble.  I think I found the solution,  there is a standby temperature setting and it looks like you have to remember to zero those if you are only slicing with one extruder.  Haven’t used cura in a long time so not sure when that became necessary.

 

Jeff.

GregValiant, 2022-08-19:
With "Extruders Share Heater" selected then Cura should not drop the hot end to the Standby Temperature.  That's kind of the point of the setting to keep the hot end at printing temperature no matter which extruder is working.  If enabling it didn't correct the problem use the "File | Save Project" command and zip and post a Project File here.  Maybe there is something else going on.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant