Ultimaker/Cura #13062: "Blocker => Modify settings for overlap" does not work for Fuzzy Skin enabled

Ic33s, 2022-08-18:
### Application Version

5.1.0

### Platform

Windows 10 64bit

### Printer

Ender 3 pro

### Reproduction steps

1. Load model
2. Enable fuzzy skin globaly
3. Add blocker
4. Changed Blocker to mode "Modify settings for overlap"
5. Added settings "Fuzzy Skin"
6. Deselected "Fuzzy Skin" checkbox

(also tried global "Fuzzy Skin" off and Blocker "Modify settings for overlap" with selected "Fuzzy Skin" on)

### Actual results

Still everything of the model was sliced with fuzzy skin.

### Expected results

Blocker "Modify settings for overlap" should have no fuzzy skin, but should be sliced "normal"

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_N0.4_-_PETG_V3_Home_Tools_HandleForSmallTools_Bottom_0h40min.3mf.zip](https://github.com/Ultimaker/Cura/files/9376919/CE3PRO_N0.4_-_PETG_V3_Home_Tools_HandleForSmallTools_Bottom_0h40min.3mf.zip)

GregValiant, 2022-08-18:
Thanks for the report.
Did you try using a Cutting Mesh instead of Infill Mesh?  As a Cutting Mesh with the only selected setting "Fuzzy Skin" and with it disabled it worked correctly for me.
Let us know if that works for you.  I'll leave this open so you can respond but it appears to be a settings issue.  I'll remove the bug label.
[image]

Ic33s, 2022-08-19:
Sorry was my fault. Thank you very much for your help.

closed 2022-08-19 by Ic33s