Ultimaker/Cura #13063: Where is the X-Ray View?

leoheck, 2022-08-18:
### Application Version

5.1.0

### Platform

Linux

### Printer

I dont have one

### Reproduction steps

Load a file, it complains about something saying to use X-Ray view. But there is no X-Ray view.

### Actual results

there is no x-ray view

### Expected results

open the x-ray view

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Hello, there is no X-Ray view

Using Cura 5.1.0
On Ubuntu 22.04

[image]

leoheck, 2022-08-18:
Learn more link throws me here:
https://support.ultimaker.com/hc/en-us/articles/360014055959

jellespijker, 2022-08-18:
https://support.ultimaker.com/hc/en-us/articles/360011733980-The-Ultimaker-Cura-workflow-explained 

This article explains how you can view the model in the x-ray view.

Does that help, or is the view actually missing?

leoheck, 2022-08-19:
> Does that help, or is the view actually missing?

I could find it now. It is hidden inside the View type dropdown list after slicing the part. Not quite easy to find when cura throw the popup that the model has errors while opening it.

fieldOfView, 2022-08-19:
I very much agree that the X-ray view is hidden too well. In the SidebarGUI plugin, I have given it a much more prominent place. In the sidebar gui, the X-ray view is back as a first class citizen in the Prepare tab, reachable via a legend similar to the line types legend in layer view.
[image]

jellespijker, 2022-08-23:
@lorenzoromagnoli maybe you have an opinion about this?

drwho83, 2024-02-28:
> I very much agree that the X-ray view is hidden too well. In the SidebarGUI plugin, I have given it a much more prominent place. In the sidebar gui, the X-ray view is back as a first class citizen in the Prepare tab, reachable via a legend similar to the line types legend in layer view. [image]

Thanks! I was going nuts trying to find it..!