Ultimaker/Cura #13064: Not sure

acedra, 2022-08-19:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Loaded original stl file. Chose selections for settings. tried to slice.

### Actual results

generated error. suggested posting bug

### Expected results

file to be sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9378539/cura.log)
[bug.zip](https://github.com/Ultimaker/Cura/files/9378547/bug.zip)

GregValiant, 2022-08-19:
Thanks for the report.
It sliced right away for me.  Someone from the Cura Team will take a look at the log file.

Vandresc, 2022-08-19:
I also get the slicing error with the attached project, I will bring it up with the rest of the team.  Thank you for the feedback.

rburema, 2022-08-19:
Tested the initial .zip the OP posted on an early alpha build of 5.2 -- slicing succeeds now. However, I'm on Linux (at the moment). Not closing yet, as @GregValiant may also be on Linux.

@Vandresc -- Where you on Windows when you tested this? Can you try with one of the 5.2 (later) alpha builds?

GregValiant, 2022-08-19:
@rburema I'm running Windows 10.

Vandresc, 2022-08-22:
@rburema yes, on Windows, and I can still reproduce with the latest alpha of 5.2.

rburema, 2022-08-23:
Ok, let's make a separate internal ticket then -- maybe we should attach the others as well if they're still failing on Windows.

jellespijker, 2022-08-23:
@rburema can you change the title of this issue in a more descriptive one?

acedra, 2022-08-23:
Sorry about that folks. First time submitting a bug to anything so I didn't
know what format to follow.

rburema, 2022-08-24:
@acedra That's fine, we get that a lot. I think @jellespijker meant for me to change the title.

jellespijker, 2022-08-24:
@acedra we welcome these reports, so please keep on submitting any issue you find or feature requests you have.