Ultimaker/Cura #13066: (5.1.0) Slicing failed

Epicurusismyhomeboy, 2022-08-19:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Reduced model that I have printed at full size in earlier version of Cura to 85%
2. Attempted to slice

### Actual results

Slice progress bar stalled and error window popped up indicating slice failed. "Slicing failed with an unexpected error."

### Expected results

Model should be sliced and ready for export. 
[cura.log](https://github.com/Ultimaker/Cura/files/9383022/cura.log)
[https://we.tl/t-pV1msPLabn](url)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Slicing seems to be much slower than in previous versions.

GregValiant, 2022-08-19:
Thanks for the report.
I'm not good at reading log files but I'm getting used to seeing this line:
**"Backend exited abnormally with return code 3221225477!"**
We could use a project file to check the model.  Load the model and prepare Cura to slice and then use the "File | Save Project" command to create a 3mf project file.  Zip the file and post the zip file here.
This appears to be a duplicate of several other reports.

Epicurusismyhomeboy, 2022-08-19:
> Thanks for the report. I'm not good at reading log files but I'm getting used to seeing this line: **"Backend exited abnormally with return code 3221225477!"** We could use a project file to check [...]

I linked the zipped project file per the bug report instructions in the original report. It's too large to upload here.

GregValiant, 2022-08-20:
I needed to copy and paste the text into Chrome as the link was a dead end.
This is a duplicate of a previous bug report.  The model appears to be fine and won't slice when it is in the center of the Cura build plate.  I read somewhere that this behavior will be fixed in 5.2.

In the meantime you can move the model on the Cura build plate by at least 10mm in the X and 10 in the Y and then rotate the model 90° clockwise.  That's how I got it to slice.
Cura 5.x does appear to slice slower.  This is a 77mb very intricate model.  I walked away from my computer after 30 minutes and it wasn't done slicing yet.  This is my new record holder for the longest to slice.  This is your 85% model.
[image]

Epicurusismyhomeboy, 2022-08-20:
Thanks for the workaround. I look forward to 5.2!

closed 2022-08-20 by Epicurusismyhomeboy