Ultimaker/Cura #13067: Cura 5.1 Z-hop error preventing slice

kd5crs, 2022-08-19:
### Application Version

5.1

### Platform

Windows 10

### Printer

Qidi X-Pro

### Reproduction steps

1. Default z-hop setting was 10mm/s, shown in red
2. Clicking Slice returns error "Unable to slice with current settings. The following settings have errors: Z Hop Speed."
3. I checked the printer json, and max Z speed is 5mm/s.
4. Changed Cura Z-hop speed to 5mm/s, setting no longer shows as red.
5. Slice still returns same error.
6. Installed Printer Setting plugin, verified it shows 5mm/s is the max just like the json.
7. Tried again with 0,1,2,3,4,5 mm/s, all same error.
8. Closed Cura, restarted, imported models again, same error.

### Actual results

5mm/s seems to be defined correctly as the max z hop speed, but Cura keeps giving that error when slicing no matter what z hop speed is selected. This happens whether z hop is enabled or not. Cannot slice.

Switching to my other printer (Kingroom KP3S), it works fine, so logically it must be a printer setting issue. Where else should I check?

### Expected results

Slices.

Thanks!

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9383284/cura.log)
[xpro cubes.zip](https://github.com/Ultimaker/Cura/files/9383285/xpro.cubes.zip)
[definitions.zip](https://github.com/Ultimaker/Cura/files/9383287/definitions.zip)

kd5crs, 2022-08-19:
Solved by changing the max in the json to 10 ("machine_max_feedrate_z": { "default_value": 10 },), then Cura sliced with a value of 5 in that field. I guess Cura was stuck a bit after the first slice error? Thanks.

GregValiant, 2022-08-19:
Nice job figuring it out.
Remember that you have made a local fix and when you next update Cura you will need to fix it again.
Can I close this?

kd5crs, 2022-08-19:
Affirmative, thanks.

closed 2022-08-19 by kd5crs