Ultimaker/Cura #13072: Built-in support inhibitor in .3mf file is rendered as solid object

rollingdice, 2022-08-19:
### Application Version

5.1.0

### Platform

Mac OS 12.5 (Apple Silicon)

### Printer

Artillery Sidewinder X2

### Reproduction steps

1. Put a .3mf object with build-in support inhibitor on build plate.

### Actual results

The model is put on the build plate but the support inhibitor is rendered as single big block on the model, like this:
[image]

### Expected results

The model should be rendered correctly, with correct positioning of support inhibitor.
Here is what it looks like in PrusaSlicer:
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is the model that I was going to print: [Outlet-Cover-ABS.3mf.zip](https://github.com/Ultimaker/Cura/files/9384630/Outlet-Cover-ABS.3mf.zip)

GregValiant, 2022-08-20:
Thanks for the report.
That 3mf file is a model file and so all objects within it are treated strictly as models.  If it had been saved out of Cura as a "Project File" then the support blocker would have imported correctly.

In Cura go to the "MarketPlace" and install the Mesh Tools plugin.  Restart Cura.
Load and select your double model.  Right click and select "Mesh Tools" and then "Split Model into Parts".  You should end up with two pieces but they are both "models".
Select the support blocker and in the "Per Model" settings you will see that it is listed to print as "Normal Model".  Change that to "Don't Support Overlaps".
This doesn't look like a bug but rather the file format chosen when that 3mf was created.  Let me know if that is correct.

MariMakes, 2022-08-26:
@GregValiant is totally right. Thanks, Greg!

If you want to learn more about support blockers in Cura, you can read about them here: 
https://support.ultimaker.com/hc/en-us/articles/360012869379-How-to-block-support-generation-in-Ultimaker-Cura

If your issue is resolved, can you let us know so we can close it?

closed 2022-09-03 by github-actions[bot]