Ultimaker/Cura #13073: Wireframe overlay rendering option

jdgarrett, 2022-08-19:
### Is your feature request related to a problem?

Sometimes it can be difficult to see details in a model at certain angles since the rendering of the model can be fairly flat, especially when the material color is black.  This can make it difficult to line up some parts for dual material rendering.

### Describe the solution you'd like

It would be nice if there was a rendering option under the "View" menu to enable a wireframe overlay of objects.

### Describe alternatives you've considered

One alternative that has helped is to change the material color to something lighter.  That helps the shading not appear as flat, but the issue still persists at some angles where a lot of faces are parallel but at different depths in the scene.

### Affected users and/or printers

This change will benefit anyone that wants to see the object wireframe for purposes of lining up multiple objects, or just to see the triangles that make up the object without having to rely on a third party modeling tool.

### Additional information & file uploads

What an object looks like with a black material:
[image]
The faces are difficult to see since shading doesn't seem to work properly for the black material.

Proposed rendering option:
[image]
In the case of the black material, the wireframe could be rendered as white to make the edges clearly visible.

MariMakes, 2022-09-22:
Hey @jdgarrett,

Welcome to the Ultimaker Github 🚀 
Sorry it took us a while to get back to you 😞 

This is a very cool request. 
I'll bring it up to the team to see what they think

Fingers Crossed 🤞

MariMakes, 2022-09-28:
Hey @jdgarrett,

I brought your request up to the team and implementing this is more difficult than I expected. 
We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand 🙏

CWempe, 2023-10-03:
Yes, I think this can improve how you use Cura!