Ultimaker/Cura #13075: Suddenly the preview is all messed up

mrelwood, 2022-08-20:
### Application Version

5.1.0-beta+327

### Platform

macOS 12.3.1 Monterey

### Printer

Creality Ender 3 V2

### Reproduction steps

- Opened any model directly from Fusion 360, or an .stl from disk.
- Slice & Preview.

### Actual results

The preview model is all messed up with non-continuous lines and coloured sectors reaching out well past the print area.

### Expected results

I expected it to work the same it did up to yesterday, which is to show a reasonable preview image and continuous printing lines.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Dynamic grapics card switching doesn't make a difference to the issue. Neither does switching the Cura print profile. Nor choosing another .stl, even those that used to work well up until today.

[image]

[Cura bug.3mf.zip](https://github.com/Ultimaker/Cura/files/9386030/Cura.bug.3mf.zip)

[Riihimökin vessan peilikaapin kahva.stl.zip](https://github.com/Ultimaker/Cura/files/9386036/Riihimokin.vessan.peilikaapin.kahva.stl.zip)

GregValiant, 2022-08-20:
Thanks for the report.
This behavior has been noted previously.  I've never been able to duplicate this on my Windows 10 laptop and I recall that it was specific to certain hardware systems (rather than operating systems).  I'll leave this as a bug report and mark it as a duplicate.  Maybe one of the developers can help out.

mrelwood, 2022-08-20:
Thanks. I was unable to find related issues before reporting.

What is strange is that I've been using the exact version of Cura extensively for a good while now without issues. Now the issue appeared out of nowhere, without doing any software or system updates or anything.

mrelwood, 2022-08-20:
Oh. Just as fast as the issue came, it now went away after simply waiting for 7 hours. I can no longer reproduce the issue. Really strange.

GregValiant, 2022-08-20:
That's the nature of this bug.  It's wiggly and slippery and without being able to duplicate, it fixing it isn't possible.   Of course it may not be a Cura problem, but whether it is or not there isn't much that can be done.
I'll close this then since once again this problem has escaped.

closed 2022-08-20 by GregValiant