Ultimaker/Cura #13076: Z seam "shortest" placement is obviously not shortest

daniels220, 2022-08-20:
### Application Version

5.1.0

### Platform

macOS Big Sur

### Printer

Creality CR-10S

### Reproduction steps

Printing an internal thread with a thin wall, such that at the minor diameter of the thread, the wall is about 2.25mm thick, while at the major diameter of the thread it is about 1.5mm thick. I am using a 0.5mm line width, so this translates into four lines (two of which are oversize) and three lines (and no way to make four fit without them being unreasonably tiny).

When I print this with a user-specified Z seam, I get a lot of stringing gumming up the threads near the Z seam, as the printer is making fast, relatively-inaccurate travel moves that seem to end up passing over open space and/or disturbing already-placed wall lines. But these travel moves could be almost entirely eliminated with a better choice of Z seam position, so I thought I'd try "shortest". Unfortunately this really does not help, it doesn't seem to be meaningfully trying to minimize the travel moves.

### Actual results

With Z seam set to shortest, the Z seams on the outside of the part are all vertically aligned, even though this results in very different lengths of travel move on each layer:

[image]

The Z seams on the _inside_ seem to "follow" the thread around the circle, but this is actually due to the seam corner preference setting—with that set to hide, hide-or-expose, or smart hiding, Cura picks up on the "corners" created where the thread appears and disappears, which is a better position than always in the same place, but nowhere near optimal. With it set to none, though, the inside seams are also in a vertical line (subject to the available vertices in the mesh):

[image]

The result is that some layers are pretty good:

[image]

While others are absolutely awful:

[image]

### Expected results

The ideal Z seam placement would be for every layer to look like this:

[image]

Where the white-and-blue boxes represent the Z seam locations. Then the only travel move needed would be shifting around the circumference at the end to the new transition point for the next layer (with the next layer looking exactly like this one, except rotated like 10 degrees). Though, actually we would need to alternate which _end_ of the single line the Z seam is at—but that's fine. Point is, there is an obvious shortest path here and Cura doesn't seem to be even trying to take it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here's a project file for the threaded part I was working on: [Thread z-seam issue.3mf.zip](https://github.com/Ultimaker/Cura/files/9386174/Thread.z-seam.issue.3mf.zip)

MariMakes, 2022-08-26:
Hey @daniels220,

Thanks for the extensive report! 🤩 

We are very aware that the quality of the Zseam has decreased since the introduction of the Arachne Engine. 😞 
It's something we really want to improve in the short term. 

We have been gathering similar reports in https://github.com/Ultimaker/Cura/issues/11877.
I'm considering this a duplicate and will be closing it in favor of the other ticket. 
You can continue following the progress here: https://github.com/Ultimaker/Cura/issues/11877

closed 2022-08-26 by MariMakes

daniels220, 2022-08-26:
Yeah, this certainly isn't the only Z-seam issue I've seen, but I was pretty sure you're already aware of the general "fuzziness"/erratic behavior, and sure enough that's what #11877 is talking about. I hope my report helps with this particular part of things.