Ultimaker/Cura #13077: Perimeters not considered bridges over support interfaces but normal fill is.

fauks, 2022-08-20:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

1. create tree support structure (tree or normal) that supports a part of a layer that has both normal fill and perimeter lines that go over it. 
2. set bridge settings and used shorter bridge line distance setting.
3. printed object

### Actual results

When printing the perimeters, it does not use bridge settings. Once it gets to the fill portion over the support structure, it starts using bridge settings (increased fan, flow settings, speed) 

### Expected results

Perimeters while going over a support area with a Z gap between the support and the current layer, should be treated as bridge settings just like the fill is.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Areas highlighted are not treated as bridge. The fill in the middle of highlighted area is.

[image]

smartavionics, 2022-08-20:
See https://github.com/smartavionics/Cura/releases for a cura variant that does what you are suggesting...

[image]

[image]

MariMakes, 2022-08-26:
Hey @fauks,

Thanks for your report 👍 
Interesting proposal, I personally haven't considered defining the layer above support to be a bridge. 
I'll bring it up to the team. 

In the meantime can you share your .3mf? 
That will help with troubleshooting. 🔍

fauks, 2022-08-26:
> 

Sure. It is in this file for the Ikea Lack v2 enclosure corner pieces [here](https://media.printables.com/media/prints/3673/stls/158821_5e268306-21c6-4a44-acdb-117fd7946322/prusa_mmu_enclosure_1.3mf)

Setting the area above a support as being a bridge would really help keep the area touching the support from adhering as well which would make removing easier, at least I would think.

fauks, 2022-08-26:
> See https://github.com/smartavionics/Cura/releases for a cura variant that does what you are suggesting...
> 
> [image]
> 
> [image]

Interesting. Too bad this hasn't been added to version 5+