Ultimaker/Cura #13078: OSX File dialogue opens on wrong desktop

stoobyct, 2022-08-20:
### Application Version

5.0.1

### Platform

OSX

### Printer

Ender 3

### Reproduction steps

Following slicing, 'Save to disk' (with multiple desktops avalable). Also does same with save to removable disk.

### Actual results

The File dialogue only shows the file type chooser and opens in a different desktop to that which cura is running in (see screenshot). Can be fixed by clicking on the workspace background with the dialogue showing or dispaying desktop list briefly. There may be other workrounds but general the file dialogue seems to need a refresh.

[image]
 

### Expected results

File dialogue should open in the Curea workspace

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Didn't do it on Cura 4.x

stoobyct, 2022-08-26:
> Hey @stoobyct,
>
> Welcome to Github 🎉 , thanks for this report.
> This looks like a bug 🐛, I'll bring it up to the team.
>
> In the meantime...
> Do you see this often or was this a one-off?
> Does it show the expected behavior if you resize the window?
>
> If it's resolved, can you let us know so we can close the issue? Thank you!

Hi,
Thank you for your reply. In answer to your questions, I have seen it frequently since upgrading from 4.13 to 5.1 (I didn’t go to 5.0) which was the main motivator to report. I didn’t try resizing the window so can’t comment. 
However, since then, I have been unable to reproduce it despite trying many combinations. I think it may be something to do with ‘assign to’ desktop options as I use multiple desktops. This was the only thing I found that was different to what I expected when I started trying in earnest to find the cause after reporting. Maybe something was not set correctly at install time which subsequent change sorted out, after which it behaved properly even when set back - I am just guessing here. If it occurs again and/ or I can reproduce it, I will be in touch further.

Kind regards

MariMakes, 2022-08-30:
Hey @stoobyct,

I took it up to the team. We believe we will not work on this anytime soon and will defer this issue to focus on more important bugs and features or until we have solid steps to reproduce. 😞

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand 😇