Ultimaker/Cura #13079: falha no fatiamento

fabioruizac, 2022-08-20:
### Application Version

5.1.0

### Platform

windos 10

### Printer

two trees bluer v3

### Reproduction steps

Abri o arquivo stl e mandei fatiar, com as minhas configurações habituais.
ai o sistema me reportou falha no fatiamento.

### Actual results

simplesmente nada, pois o programa não esta conseguindo fazer o fatiamento

### Expected results

que faça o fatiamento e de a previa das camadas.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

o pagina não suporta o arquivo stl

GregValiant, 2022-08-20:
Thank you for the report.
Without a Project File there can be no investigation.  Use the "File | Save Project" command to create a 3mf file with your model and settings in it.  Add it to a Zip folder and post the Zip folder here.

fabioruizac, 2022-09-04:
Ok