Ultimaker/Cura #13080: One perimeter top and bottom layer

PuszInPusz, 2022-08-20:
### Is your feature request related to a problem?

Not really a problem but a way to improve the quality and the looks of the printed parts. There are two features called one top perimeter and one bottom perimeter in Superslicer and it really makes the parts look more polished and professional, even without ironing. Only the top most (in Cura called top skin) perimeter is affected by this setting, this really being a quality improvement.

### Describe the solution you'd like

I’d like cura to implement this feature because it really makes everything look better and more feature rich (most advanced 3d printing software 😉). I myself am a heavy Cura user (using since 4 years) but the one perimeter setting switched me to Superslicer because it is really awesome and I’d like to switch back to cura for the reliability and tons of features but I thinks this setting just is a very big reason to using Superslicer.

### Describe alternatives you've considered

No alternatives 

### Affected users and/or printers

Every user will benefit and I think stuff like printed lithophanes can get a quality boost and maybe win some new user who just use Superslicer for this feature alone like myself 

### Additional information & file uploads

[image]

5axes, 2022-08-22:
I love this parameter in SuperSlicer could be a nice improvement of your Cura Master @smartavionics ?

smartavionics, 2022-08-22:
My immediate conclusion is that it will be quite hard to implement in Cura but I will give it some more thought...

PuszInPusz, 2022-08-24:
That sounds great, why would it be hard though? I’m not a programming professional but can’t you just like see what SS does and kinda copy it, like they did with the Arachne engine xD

smartavionics, 2022-08-24:
Sure, you can observe what SS does (I have already) but the devil is in the detail of the implementation. For Cura it's a chicken and egg situation because it currently determines where the walls are first and then fills with skin (and infill). But for this feature you need to alter the processing order to be something like: 1) calculate shape of outer wall, 2) calculate shape of skin areas that have air above, 3) calculate shape of any inner walls so that they hug the outer wall and the areas of skin that have air above. Also, in a quite a few places in the Cura code, it assumes that the walls all lie next to each other but that isn't true when this feature is used. So it's not trivial to implement (at first thought!)

PuszInPusz, 2022-08-24:
Oh okay, that sounds like a lot. Well I hope someday (soon if possible) the feature will be a standard in cura.

5axes, 2022-08-24:
Thanks for the explanation @smartavionics . It's informative and it's also often something that's missing from Ultimaker's responses. Many requests end up in the "Deferred" category because they certainly represent a lot of work, but without an explanation it is difficult to know if the request represents 1 hour or 1 day or 1 month of development.

smartavionics, 2022-08-30:
I'm making progress and hope to have this feature in my next release (not yet scheduled). Still got some problems to sort out.

[image]

PuszInPusz, 2022-08-30:
Wow!! That’s what I call support. Taking ppl needs and requests seriously and starting implementing them. I’m very hyped already and can’t wait to see if it’s going to workout and how it will look. I’m very happy that you really considered my request. Wish you all the lucky!! And thanks a lot.

smartavionics, 2022-08-31:
Nearly ready...

[image]

PuszInPusz, 2022-08-31:
If it’s working and stuff, how long would it take until it gets public? Or is there a way to get the alpha version so that I can do test and maybe give feedback and such?

smartavionics, 2022-08-31:
OK, I'll make a release today so you can give it a go. Should have it done in a few hours.

smartavionics, 2022-08-31:
I've released 4.20.4 over at https://github.com/smartavionics/Cura/releases. If you try it out and have any feedback, please create an issue at https://github.com/smartavionics/Cura/issues rather than here, thanks.

[image]

PuszInPusz, 2022-08-31:
Thank you a lot!! I’m soo happy to try it out 🥲

5axes, 2022-08-31:
Not an issue just a question : https://github.com/smartavionics/Cura/issues/183

PuszInPusz, 2022-08-31:
I just started a print, I'm very curious how it will turn out.