Ultimaker/Cura #13081: Error while slicing complex STL

KameSense, 2022-08-20:
### Application Version

5.1.0

### Platform

Kubuntu 21.04

### Printer

Wanhao D12

### Reproduction steps

Load the model, resize to 80%, slice

### Actual results

Error while slicing.
No error with Cura 5.0.0

### Expected results

Sliced model :)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://we.tl/t-DhtpfqEye0

GregValiant, 2022-08-20:
Thanks for the report.
The resolution of the STL file is extremely high and that accounts for the 312mb file size.  The STL file also has a considerable number of errors in it.  It is likely that all those errors are keeping Cura from being able to slice it.  When you scale the model down the triangle density goes up and that may be part of the problem as well.
I attempted to repair the model with MS 3D Builder.  At this point it's been trying for 50 minutes..  I think I just don't have enough horsepower in this computer to deal with the file.  Someone from the Cura Team will take a look.

KameSense, 2022-08-20:
Yes, I'm aware of all these issues, but Cura 5.0.0 is still able to slice it successfully though :)
Intel(R) Core(TM) i7-12700K with 32Gb of RAM and nVidia GTX 1650 here

MariMakes, 2022-08-26:
Hey @KameSense,

I gave it a go, but also my computer is objecting and I lost my patience. 
Are you sure you want to 3D print this? 🙈 

The model is really big and I even get messages that it's broken. 
[image]

I'll bring it up to the team, but I'm worried that it might not get any priority since it's a complicated model.

MariMakes, 2022-11-08:
Hey @KameSense,

To me, it doesn't seem like a bug in Ultimaker Cura. The mesh that you are trying to slice is faulty. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2022-11-08 by MariMakes