Ultimaker/Cura #13083: Cura clshed during slicing

roameeo, 2022-08-20:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality Ender 5

### Reproduction steps

1. Loaded 3 STL files
2.  Positioned them how I wanted them
3.  Clicked slice

### Actual results

It started slicing then crashed.

### Expected results

Should produce g-code for the project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9387953/cura.log)
[Dragon_Right_Wing_Fingers_Full.zip](https://github.com/Ultimaker/Cura/files/9387961/Dragon_Right_Wing_Fingers_Full.zip)

GregValiant, 2022-08-20:
Thanks for the report.
I was able to slice your 3mf file with no changes to it so .I am unable to duplicate the problem.  Someone from the Cura Team will check your log file.

roameeo, 2022-10-11:
Thank you! I was using a couple of the experimental settings. I wonder if that had something to do with it?

I removed one of the STL's and was able to slice it, so I'm good to go.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker