Ultimaker/Cura #13086: Failed to slice

BoogerX, 2022-08-21:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Creality Ender 5Pro

### Reproduction steps

Cura will not slice model.

### Actual results

Still will not slice

### Expected results

??

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5_Baby_Groot_Planter.zip](https://github.com/Ultimaker/Cura/files/9389247/CE5_Baby_Groot_Planter.zip)

MariMakes, 2022-08-26:
Hey @BoogerX,

Welcome to Github and for thank you for your report 🎉 
I'm also having a failed slice with your .3mf but I can successfully slice it with the default settings. 
So it might have to do with the combination of a few settings. 

[image]
[image]

We'll have to do some closer investigation. I'll keep you up to date 💪

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker