Ultimaker/Cura #13090: New 2 Walls Adaptive Line Width Feature!

cemalcraycle, 2022-08-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

We are building 3D Printed RC planes as Craycle Hobby. Our models has 2 walls on all model body. We makes models walls 0,84 mm to get continues 2 walls printing. Some model walls could be 1, 1.5 or 2,5 mm and we make this for some of mechanical design reasons.   

### Actual results

When the model walls transitions from the 1mm+ to the 0,84 mm, walls are sticking early together and gaps are between walls are filling by making thicker walls automaticly. This feature comes with Cura 5.1.0 and this situation brokes all of our model designs mechanical approach.

### Expected results

We are sharing our Cura Profile so people can print our models perfect without adjustments. We want to keep Cura updated so people can continue to use our Cura Profile who use updated Cura Slicer. Probably this feature makes better for solid printings but all of our models can't be printed with Cura 5.1.0. We are suggesting this feature can be made optional in Walls Settings Tab. There is tons of products in the market which have 2 walls some of areas. They are going to be bad too. 

I am attaching a few sample photos to illustrate the problem. All the lines you see here should not be in the model.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I am attaching a few sample photos to illustrate the problem. All the lines you see here should not be in the model
[image]
[image]
[image]
[image]
[image]
.

MariMakes, 2022-08-26:
Hey @cemalcraycle 

Those are some cool models ❤️ 

Cura 5.x introduces a new way to build up walls. 
Super thin-walled models will probably need a little more tuning but overall we see an increase in Printed Part Quality.

Have you checked out the newly introduced settings yet?

cemalcraycle, 2022-08-26:
@MariMakes I am glad you like my models. This is awesome feature that I calling Adaptive Line Width Feature. I read your article and its looks like the same as I thought but It's so diffucult to tell to our customers this type of major updates. I made all of my models same profile. All of the model parts printing with the same profile. Just some little parts need top layers and we just add 2 top layers them. So it's so important Cura hasn't updated with like this major features which is made default.

We wish this feature was in the experimental section. In our models we don't want to print adaptive line width, we wants to get straigth 0,4 mm layer width on everywhere. Some areas has 1 mm wall thickness to get 0,2 mm space between walls. Now it's so hard to control this.

Ultimaker Cura gives us freedom to educate to our customerses how to print our models quick. It's so important that there is a free Slicer in the market for whole world! and we are so happy to see it's continue to improving. Congragulations for the new awesome features!