Ultimaker/Cura #13091: Importing Cura Profile Problem!

cemalcraycle, 2022-08-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

We are building 3D printed RC planes and sharing our Curaprofile to people can print our models without settings adjustments. There is critical setting in profile to ensure wall printing fails. 

### Actual results

Appart from the Version, always we get error while importing Cura Profile. We found a way to make it succes; Creating new custom printer. When you create a new Custom Printer, finally you can import our craycle.curaprofile.

### Expected results

Sometimes creating new Custom Printer can't solve this problem so is there any solution for this?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

 Here is profile link: https://1drv.ms/u/s!Au6X2Jil-WVDgdoLauI7RhBbm4dRgg?e=ePXLdN

MariMakes, 2022-08-30:
Hey @cemalcraycle 

I changed your bug to a feature request.
Sharing settings between completely different printers is a functionality that we currently do not have. 

We do have some people in the community that have found different solutions for it like 5Axes with [the Import Export CSV plug-in ](https://marketplace.ultimaker.com/app/cura/plugins/5axes/ImportExportProfiles) or [Export HTML Cura Settings plugin](https://marketplace.ultimaker.com/app/cura/plugins/5axes/HTMLCuraSettings) 

I've also seen users sharing their settings via .3mf project files. They would load a random cube shape and ask others to load the project file, remove the cube and load the model they want to slice on the build plate. But also this solution does not work for different printers. 

Would these workarounds be applicable to your situation?

cemalcraycle, 2022-08-30:
@MariMakes,

Hello,

I saw these plugins last week and thought about it. Sharing screenshots of the settings still seems more effortless for the customers. I do not share screenshots of the settings, I want to teach them importing profile but our customers who cannot import curaprofile constantly demand this. They don't want to use the method we found for successful import. Adding a new printer ensures successful import and we describe this in a separate section in the User Guides of all our products. It will be necessary to explain the use of plug-ins in the same way or by shooting a Youtube video. I don't think there will be enough recovery for people. These plugins also can sometimes be removed. I can no longer find a useful plugin in the store that I saw in a video.

This is an important issue for all 3D Model Sellers or Providers. Designers should be able to share their settings with people very comfortably. Some of our customers don't even know that the settings can affect the pressure, it is very difficult to explain the profile mortgage to this segment. These people are meeting with the 3D Printer with our models. 

Technically I can partially understand the profile needing to add a new printer. If it is possible; A setting profile file, completely independent of the printer and material properties, can be produced in which only the settings are transferred, and this file can be imported in the profiles section. If this is not possible, curprofile should be improved. This is an important feature to use now and in the future. 

We can always find a way to share our cura profile, such as sharing screenshots, but it's all about the convenience and ease of using Cura Slicer.

5axes, 2023-03-09:
Why don't you create your Own Plugin , You will be able to automatically analyse the printer configuration and modify the customer settings.  Something like the UniversalCuraSettings Plugin : https://marketplace.ultimaker.com/app/cura/plugins/5axes/UniversalCuraSettings