Ultimaker/Cura #13093: Trying to run from source on windows 10

phedg1, 2022-08-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker Doesnt Matter

### Reproduction steps

Install the dependencies for windows 10.

Uninstall conan using pip.
Remove .conan directory from user directory.

Clone the main branch. Or clone the 5.1 branch. Both have the same issue.

Open powershell as an admin in the cura source root directory.
Run:
pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
conan install . cura/5.1.0@ultimaker/stable --build=missing --update

### Actual results

mpdecimal/2.5.0: Configuring sources in C:\.conan\2d80c688489c3b7\1
Downloading mpdecimal-2.5.0.tar.gz completed [2499.97k]

mpdecimal/2.5.0: Copying sources to build folder
mpdecimal/2.5.0: Building your package in C:\.conan\ac27e231edc3d\1
mpdecimal/2.5.0: Generator 'MSBuildDeps' calling 'generate()'
mpdecimal/2.5.0: Generator txt created conanbuildinfo.txt
mpdecimal/2.5.0: Aggregating env generators
mpdecimal/2.5.0: Calling build()
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.2.5
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

----Running------
> "C:\.conan\ac27e231edc3d\1\conanbuild.bat" && nmake /nologo /f Makefile.vc libmpdec-2.5.0.lib MACHINE=x64 DEBUG=0 DLL=0 CONAN_CFLAGS="-O2 -Ob2 -MD" CONAN_CXXFLAGS="" CONAN_LDFLAGS=""
-----------------
Capturing current environment in C:\.conan\ac27e231edc3d\1\deactivate_conanbuildenv-release-x86_64.bat
Configuring environment variables
        cl /W4 /wd4200 /wd4204 /wd4221 /D_CRT_SECURE_NO_WARNINGS /nologo /DCONFIG_64 /DMASM /Ox /GS /EHsc -O2 -Ob2 -MD -c basearith.c
basearith.c
basearith.c(29): fatal error C1083: Cannot open include file: 'mpdecimal.h': No such file or directory
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.32.31326\bin\HostX64\x64\cl.EXE"' : return code '0x2'
Stop.
mpdecimal/2.5.0:
mpdecimal/2.5.0: ERROR: Package '5c8fc3ef602a8e57d81400f2865131af6a719c3d' build failed
mpdecimal/2.5.0: WARN: Build folder C:\.conan\ac27e231edc3d\1
ERROR: mpdecimal/2.5.0: Error in build() method, line 154
        self._build_msvc()
while calling '_build_msvc', line 99
        self.run("""nmake /nologo /f Makefile.vc {target} MACHINE={machine} DEBUG={debug} DLL={dll} CONAN_CFLAGS="{cflags}" CONAN_CXXFLAGS="{cxxflags}" CONAN_LDFLAGS="{ldflags}" """.format(
        ConanException: Error 2 while executing nmake /nologo /f Makefile.vc libmpdec-2.5.0.lib MACHINE=x64 DEBUG=0 DLL=0 CONAN_CFLAGS="-O2 -Ob2 -MD" CONAN_CXXFLAGS="" CONAN_LDFLAGS=""

### Expected results

I expected the cura build environment to be initialized successfully.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've been trying to get to get any version of cura to run or build from source since late June. But it seems that the point in time I started trying to do this was when cura was going through some kind of build process overhaul. I still haven't succeeded once. I've been trying this on the 5.1 branch and the main branch. This issue has been happening for a week or two, but since I never saw an issue created for it I assumed I must be doing something wrong and have spent the last week trying to troubleshoot it.

Anyway, there seems to be a missing file for mpdecimal 2.5.0. I can't figure out where to get it from or where to put it if I did. I hope this is an easy fix.

jellespijker, 2022-08-23:
What happens if you try the install command with:
```
conan install . cura/5.2.0-alpha@ultimaker/stable --build=missing --update
```

phedg1, 2022-08-23:
In switched to that specific Conan version and I tried the install command you have there. I got the exact same errors in the log. I cleared the cache before trying this.

phedg1, 2022-10-07:
Still getting more or less the same error, on the latest commit on the 5.1/5.1.1 branch (September 7th).

jellespijker, 2022-10-10:
can you try again using the instructions from this comment https://github.com/Ultimaker/Cura/issues/12773#issuecomment-1272992115

phedg1, 2022-10-11:
@jellespijker Winner winner! That worked perfectly, it built and ran without any further issues at all. Thank you!

I'm looking forward to the upcoming pull requests for this!

jellespijker, 2022-10-12:
great news I will close this issue as fixed.

FYI @MariMakes

closed 2022-10-12 by jellespijker

phedg1, 2022-10-21:
So, I've had a bit of time to play with this. I'm still getting more or less the same error when curaengine is in editable mode.

I've made some edits to both cura and curaengine for this. I haven't had the time yet to see if this happens without those edits. I'm hoping to do this in the next couple of days and then I'll post those logs.

Running just cura without curaengine in editable mode works both with and without the edits. I am in the latest 5.2 commit for both, building the 5.2.1 version for both.

phedg1, 2022-10-21:
@jellespijker I don't know if you'll get this on this closed issue.

phedg1, 2022-10-22:
I have created a new issue for this:
https://github.com/Ultimaker/Cura/issues/13622