Ultimaker/Cura #13094: feill to slice the file

EffectStyle, 2022-08-22:
### Application Version

5.10

### Platform

Windows 11 

### Printer

prusa mini

### Reproduction steps

i press slice i have not change anything all the other stl no problem whit the same setting 

### Actual results

start to think get to about 30 % blut and then when back to 0 after 5 min 
i close it and then try one more time in 10 min show my the report function

### Expected results

i do not know mb is just file was to big on the prusaslicer show up 12 h 
but i have try on other files that takes full day whit no problems 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9396658/cura.log)
[OPM_1553-15-_15_body_top_1_L_FDM (repaired).zip](https://github.com/Ultimaker/Cura/files/9396646/OPM_1553-15-_15_body_top_1_L_FDM.repaired.zip)

MariMakes, 2022-08-26:
Hey @EffectStyle,

Welcome to GitHub! Thanks for your report. 🎉 

I'm having some trouble reproducing it on my side. 😞 
I'm able to succesfully slice on standard prusa printers.
[image]

Can you please share your configuration file? Than I can see what settings you have for your Prusa Mini. 
If you click Help, Show Configuration Folder, you'll see a bunch of files.
Can you zip those for me and share it here. Don't forget to open Github in your browser to share files. 

Thank you! 👼

EffectStyle, 2022-08-26:
[cura.zip](https://github.com/Ultimaker/Cura/files/9434146/cura.zip)
[gamebody.zip](https://github.com/Ultimaker/Cura/files/9434187/gamebody.zip)
[MINI_cura_profiles.zip](https://github.com/Ultimaker/Cura/files/9434185/MINI_cura_profiles.zip)
i have use the default from the prusa side 
https://help.prusa3d.com/guide/how-to-import-profiles-to-cura-4-x-windows-macos_17848#17856&_ga=2.259925890.944902336.1661525381-1530490878.1648834170

MariMakes, 2022-08-29:
Hey @EffectStyle,

Thanks for the extra info and files.
I've been able to recreate the crash, so I'll bring it up to the team to analyze. 
Fingers crossed 🤞 

I also checked your settings and it seems that if you change your layerheight slightly, you'll have a succesful slice!
[image]
Hopefully you are not stuck anymore.

EffectStyle, 2022-08-29:
tanks you and the team :)

closed 2022-08-29 by EffectStyle