Ultimaker/Cura #13097: Not slicing model

TonyK54, 2022-08-23:
### Application Version

5.1.0

### Platform

MacOS

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. using extension "Part For Calibration" selected "Acceleration Tower"
2. clicked on "Slice" button

### Actual results

Received message to send bug notification

### Expected results

for model to be sliced. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[AccelerationTower.3mf.zip](https://github.com/Ultimaker/Cura/files/9401707/AccelerationTower.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9401710/cura.log.zip)

MariMakes, 2022-08-29:
Hey @TonyK54 

Love the cat profilepicture 😻 
I'm able to recreate your crash on our side. 
It's quite a quick crash so I'll bring it up to the team.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker