Ultimaker/Cura #13101: Slicing failed with no info

booms8, 2022-08-24:
### Application Version

5.1.0

### Platform

Linux Mint

### Printer

Ender 3

### Reproduction steps

1. Import specific model
2. Click "Slice"

### Actual results

Receive "Slicing failed with an unexpected error" dialogue

### Expected results

Sliced model ready for previewing and printing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[failed_slice.zip](https://github.com/Ultimaker/Cura/files/9412553/failed_slice.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9412557/cura.log)

MariMakes, 2022-08-29:
Hey @booms8 ,

Thanks for reporting this issue 🎉 

I'm seeing the crash in your logs as: [WARNING] Can't create a transition when connecting two perimeters where the number of beads differs too much! 2 vs. 0 

But when I tried to reproduce your crash on my Windows machine I had a succesfull slice.
[image]

Can you doublecheck for me if you included the correct .3mf?
Than I'll have someone from the team with a Linux device check aswell.

Thank you.

booms8, 2022-08-30:
I'm pretty sure it's the right file but I'll double check. I was able to
get the slice to work; changing quality from "Dynamic" to "Super" caused
the issue (no settings changed except the layer height), but directly
setting the layer height to 0.12 worked.

closed 2022-09-15 by github-actions[bot]