Ultimaker/Cura #13105: Preferences UI: window focus does not return to foreground

hoogman, 2022-08-24:
### Application Version

5.1.0

### Platform

macOS 12.5.1

### Printer

Creality Ender 6

### Reproduction steps

1. Open "Manage Profiles" via Profiles -> Manage Profiles or CMD+j.
2. "Preferences" window will open. Select main window. Preferences window will drop to background.
3. Select Profiles -> Manage Profiles or CMD+j to return to preferences screen.  Preferences windows does not return to foreground and remains hidden in the background.

### Actual results

Preference window does not return to foreground and remains in the background hidden behind main window.

### Expected results

When selecting "Manage Profiles" or CMD+j I would expect the "Preferences" window would be opened in the foreground if not already open or the window to be returned to the foreground if currently open in the background
[image]
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

MariMakes, 2022-08-29:
Hey @hoogman,

This looks like a 🐛 bug!
I'll bring it up with the team to see what we can do. 
I'll keep you posted 🤞

MariMakes, 2022-08-30:
Hey @hoogman,

I took it up to the team. We believe we will not work on this anytime soon and will defer this issue to focus on more important bugs and features or until we have solid steps to reproduce. 😞 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand 😇

closed 2022-08-30 by MariMakes