Ultimaker/Cura #13108: Ender 6 printer ignores material selection.

GroovyDrifter, 2022-08-24:
### Application Version

5.0.0 & 5.1.0

### Platform

Windows 10

### Printer

Creality Ender 6

### Reproduction steps

Once the Ender 6 is selected as printer, Cura ignores the material selection. It appears fixed as 220ºC Extruder and 50ºC Hot Bed

### Actual results

No change in material temperatures.

### Expected results

Material temperatures should be equal to the selected material.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

Asterchades, 2022-08-25:
This is an issue with the Ender-6 profile, which specifically configures the temperatures (220/50) rather than reading back the material properties. It also pre-sets the retraction distance (10). You can change the values in the print settings and they will work.

Ultimaker takes a "hands off" approach when it comes to 3rd party profiles, only making alterations when absolutely necessary (such as increasing the minimum resolution settings with the release of 4.8). As such this will require a community-submitted pull request for the profile to have corrected. Should you with to make such a pull request the file in question is available [here](https://github.com/Ultimaker/Cura/blob/main/resources/definitions/creality_ender6.def.json) (simply removing the temperature and retraction lines would be sufficient).

Alternatively just about any other Ender profile could be used as long as the bed dimensions are updated accordingly - though the Ender-3 profile features exclusion zones to accommodate for the bed clips.

closed 2022-08-25 by Asterchades

GroovyDrifter, 2022-08-25:
Thank you! I'll try to learn to use GIT to fix this.

GroovyDrifter, 2022-09-01:
Made a pull request, hope it is ok https://github.com/Ultimaker/Cura/pull/13111