Ultimaker/Cura #13114: Slicing failed

the-real-loondawg, 2022-08-25:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

With attached project file, changed infill overlap percentage to 10.0%. 

[CE3PRO_tidy_cat_funnel.zip](https://github.com/Ultimaker/Cura/files/9424839/CE3PRO_tidy_cat_funnel.zip)
.

### Actual results

Received an unexpected error slicing. Cura asked me to report a bug. Same file with infill overlap percentage set to 0.0% slices without error

### Expected results

Slicing with out unexpected error.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

NA

MariMakes, 2022-08-30:
Hey @the-real-loondawg,

Welcome to Github! 🚀 

I took a look at your file and I'm having the same slicing error. 
[image]
Thanks for sharing! 

I'll bring it up to the team for further investigation 🔍

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker