Ultimaker/Cura #13115: (5.1.0) Model Not Fitting Buildplate Despite Having Room

GunsOfSpuds, 2022-08-25:
### Application Version

5.1

### Platform

Windows 10

### Printer

Custom (Voxelab Aquila C2)

### Reproduction steps

1. Set build plate to 220x220x250
2. Imported stl to be printed (max dimension only 165mm)

### Actual results

Part greyed out as though it is too large for the build volume, when it can obviously fit in the build area.

### Expected results

Part slicing able to function.
[image]
[image]
[image]
[image]
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Also, by exaggerating the build volume to 400x400, the part is able to be sliced, however half of the build volume is unusable. With the 220x220 volume, the entire area is unusable. Smaller parts are able to be sliced

vwpollen, 2022-08-26:
It looks like your model has a small bit of volume at the tip of the shade. Once you clean the model to remove this, then you will obtain a shadow the size of your model and you will be able to slice it.
[image]

Have a good day.

MariMakes, 2022-08-30:
@GunsOfSpuds I'm closing this issue since it's an error in the model and not in Cura. 😄