Ultimaker/Cura #13118: Start code variable for the gcode file name

Triangulix, 2022-08-25:
### Is your feature request related to a problem?

Since I often print several variants of an object for testing purposes I want a reminder on the printer display of which file is currently printed.
Currently I'm using "M117 Printing {jobname}" in the start code but this doesn't fit my needs.

### Describe the solution you'd like

Another variable that contains the file name of the gcode file

### Describe alternatives you've considered

I searched for a Marlin feature to do that but didn't find one.

### Affected users and/or printers

Everyone

MariMakes, 2022-11-01:
Hey @Triangulix,

Sorry, it took us a while to get back to you 😞 

I'm not sure if I understand your problem because our Ultimakers display the name of the file that they are printing on the LCD screen. 

Perhaps this plug-in can help you. 
https://github.com/fieldOfView/Cura-CustomJobPrefix

Would you still like to keep this issue open or do you agree that we can close it?

Triangulix, 2022-11-01:
The problem only arrises if other M117 commands display something. At the moment I have removed all M117 commands from the start code to keep the file name on the display.

MariMakes, 2022-11-02:
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to defer this issue. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand.

Triangulix, 2023-11-03:
I don't care anymore.

closed 2023-11-03 by Triangulix