Ultimaker/Cura #13122: Slicing failed on model when using tree support structure

Angelusz, 2022-08-25:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Monoprice Select Delta Mini v2

### Reproduction steps

Tried to slice a model using support structures with settings set to tree. Without support it sliced fine (but the model would not be printable)

### Actual results

Program got stuck loading slicing, no output.

### Expected results

I expected it to show a sliced model with all the layers and information etc. -- now it just keeps 'loading' forever.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Cura gave a watertight mesh error, perhaps cura was unable to correctly interpret the model file leading to gaps in the mesh, c.q. support?

Slicing with the setting support to "normal'' instead of 'tree' does lead to a sliced model. The program runs into an error while parsing the tree support structure.

Branche angle was at 40 degrees. When I changed it to 60 degrees, it did slice correctly with tree support.

Angelusz, 2022-08-25:
https://we.tl/t-cmSbtSbrT8

MariMakes, 2022-08-30:
Hey @Angelusz,

Welcome to the Ultimaker Community 🚀 

I took a look at your impressive model 👀, but I'm able to slice it successfully. 
[image]

It's hard for us to fix an issue we don't have solid steps to reproduce.
Can you take another look for us to see if you can reproduce it or if you have shared the correct file?

closed 2022-09-14 by github-actions[bot]