Ultimaker/Cura #13123: would not slice

Brucejb, 2022-08-25:
### Application Version

5.0.0

### Platform

windows 10

### Printer

ender 6

### Reproduction steps

down loaded a file from Thingiverse but when i went to slice the file it failed and no matter what i did it wouldn't change, it always ended the same way saying "fail to print"

### Actual results

N/A

### Expected results

N/A

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

N/A

MariMakes, 2022-08-30:
Hey @Brucejb,

Welcome to the Ultimaker Github 🚀 
I want to take a look at your issue, but your project file seems to be missing.

Can you share the project file with me that shows the crash? 
To save a project file go to File -> Save project. Please make sure to .zip your project file. 
For big files you may need to use WeTransfer or similar file sharing sites.
Remember, G-code files are not projectfiles. 😄 

Thank you 😇

Brucejb, 2022-08-31:
This was the file project that I was trying to slice and print but would only error when I go to slice, it downloaded just fine though. 

https://www.thingiverse.com/thing:1948889

MariMakes, 2022-08-31:
Hey @Brucejb,

I took a look at the models. 
When I opened them on my Ultimaker S3 I was greeted with this message that the model is not watertight.
That usually means that the model is not fit for 3D printing (yet). 
You can use plug-ins like [the Mesh Tools Plugin ](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools) or other tools to attempt to fix the model. 
[image]

The model sliced for me with the default settings, and when I looked at the initial layer I could see that the model is broken and  has internal holes. 
[image]

If I were you, I would print the remix mentioned in the thing. 
https://www.thingiverse.com/thing:3871113 - The file that starts with UM2 looks way more promises and slices like this:
[image]

I'm going to close this issue since it's not an issue in Cura but an issue in the model. 
Good luck with printing!

Brucejb, 2022-08-31:
Thank you good sir 👍

MariMakes, 2022-09-01:
Happy to help! 

[image]