Ultimaker/Cura #13125: can not run cura from source

KrIsChUnn, 2022-08-26:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

I download all necessary requirements as the file asked. but when I operating as the website(https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) says, Error appeared.

### Actual results

the error message was:
ERROR: Conan v2 incompatible: Use 'self.cpp_info.build_modules["<generator>"] = ['lib\\cmake\\protobuf\\protobuf-generate.cmake', 'lib\\cmake\\protobuf\\protobuf-module.cmake', 'lib\\cmake\\protobuf\\protobuf-options.cmake']' instead

### Expected results

there should be no errors.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

please help me about it.

jellespijker, 2022-08-26:
Please pass the complete output of the `conan install` command. This is to little information for us to help you with.

KrIsChUnn, 2022-08-26:
thank you for your reply, here is the output of the conan install below.
[image]
[image]
[image]
[image]
that's all outputs.

MariMakes, 2022-08-30:
Hey @KrIsChUnn,

Just wanted to say Welcome to the Elite Running from Source Club 👑 
I can't say that I reached that tier yet 😉 

I just know that we recently updated our wiki regarding running from source.
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
Maybe it holds the answers you need. 

@jellespijker can you take a look at the output, please?

jellespijker, 2022-08-30:
Hi @KrIsChUnn did you set the `CONAN_V2_MODE` variable in your environment by any chance? If that is the case can you check what happens if you remove that variable from the Environment?
https://docs.conan.io/en/latest/reference/conan_v2_mode.html

Alternatively you can also remove the protobuf package from the local conan cache and try for it to download the latest recipe.
```powershell
conan remove protobuf -f
```

KrIsChUnn, 2022-09-02:
Hi @jellespijker ,I reinstalled  Win11 in my pc and do as the Running from Source says, it happend to be successful. However the error with PyQt6 comes after. I'm gonna to deal with it according to Issues commited by others. Thanks a lot for helping me !

closed 2022-09-02 by KrIsChUnn

jellespijker, 2022-09-04:
Hi @KrIsChUnn thanks for letting me know. Good to know that your issue has been resolved. Sad to know that you run into #12773 . But keep following that thread, we're working on that