Ultimaker/Cura #13129: No random "Z Seam Alignment" when "Fuzzy Skin" is activated

AndyS1957, 2022-08-26:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Artillery Genius / Artillery Genius Pro

### Reproduction steps

1. Activation "Z Seam Alignment" -> "Random"
2. Activation "Fuzzy Skin"
3. Printing a cylinder

### Actual results

The "Z-Seam Alignment" is not "Random", but in a vertical line on one side of the cylinder. This happens only, when "Fuzzy Skin" is activated. When not, "Z Seam Alignment" works correct

### Expected results

I expect a random "Z Seam Alignment"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[PRO_Topf_130x130_3farb_GPETG.zip](https://github.com/Ultimaker/Cura/files/9432059/PRO_Topf_130x130_3farb_GPETG.zip)

Asterchades, 2022-08-27:
Duplicate of #12190.

closed 2022-08-27 by Asterchades

nallath, 2022-09-08:
This has been fixed for 5.2