Ultimaker/Cura #13135: Question: Any Plan for support ARM architecture?

seungwook-jung, 2022-08-27:


smartavionics, 2022-08-27:
Hi, this doesn't answer your question but you may find it interesting anyway... My Cura variant (see https://github.com/smartavionics/Cura/releases) has builds for 32 and 64 bit Raspberry Pis. They obviously will not run on an ipad but now you know that Cura 4 can be built for the ARM architecture.

seungwook-jung, 2022-08-27:
@smartavionics thank for your comment!
 I think it's worth a try to make Lib for iOS 
thanks!

jellespijker, 2022-08-27:
You can also try out following the build instructions https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

But change the Conan profiles https://github.com/Ultimaker/conan-config/tree/master/profiles

With the following instructions https://docs.conan.io/en/latest/systems_cross_building/cross_building.html

I know it is at least possible at the moment to compile the engine for the M1 natively (ARM). All unofficially supported atm.

jellespijker, 2023-08-18:
I'm in the process of setting up the workflows and hooking up the physical self-hosted runner for building Cura for the M1.

**Can people try the binaries in this run, which are compiled for the native M1's:** https://github.com/Ultimaker/Cura/actions/runs/5906663040

- https://github.com/Ultimaker/Cura/suites/15242781431/artifacts/871059202
- https://github.com/Ultimaker/Cura/suites/15242781431/artifacts/871059203

I am mostly curious if these run on older MacOSes

closed 2023-08-23 by casperlamboo