Ultimaker/Cura #13139: crash

AlgenGruetze, 2022-08-27:
### Application Version

5.1.0

### Platform

Windows 8.1

### Printer

Ai3 Mega X

### Reproduction steps

start "Slice"
messed up after 10%
msg: Application failed , must be closed...

### Actual results

messed up

### Expected results

sliceing should finish

im not been payed by yas to do your work

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9438415/cura.log)
[cura-1.log](https://github.com/Ultimaker/Cura/files/9438418/cura-1.log)
[image]
[image]
[image]
[image]

jellespijker, 2022-08-27:
Please provide a 3mf with the project that is failing.

I would also want to remind you that Cura is a free product, provided by Ultimaker to the public. I think people will respond better on a request for help if it has a little less demanding tone about delivering a fix for a product which is free for a third-party printer.

MariMakes, 2022-08-30:
Hey @AlgenGruetze,

@jellespijker is right, I need a project file to start troubleshooting your issue. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files you may need to use WeTransfer or similar file sharing sites. Rememer, G-code files are not project files.

Can you share your project file with us?

closed 2022-09-14 by github-actions[bot]