Ultimaker/Cura #13140: CMakeLists.txt asks for ./cura/CuraVersion.py.in which doesn't exist anymore

JeanLucCoulon, 2022-08-27:
### Application Version

5.1.0 (git master)

### Platform

Linux

### Printer

Biqu B1

### Reproduction steps

I tried to build (cmake).

### Actual results

cmake ..
CMake Error: File /usr/local/src/cura/cura-5.1.0/cura/CuraVersion.py.in does not exist.
CMake Error at CMakeLists.txt:35 (configure_file):
  configure_file Problem configuring file

-- Linking and building  against Python 3.10.6
-- Using CURA_BINARY_DATA_DIRECTORY from set of environment variables...
etc.

### Expected results

Normally build cura.
It seems that ./cura/CuraVersion.py.in has been removed from the source tree (maybe replaced with CuraVersion.py.jinja without CmakeList.txt being updated?).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

--

jellespijker, 2022-08-27:
CMakeList.txt is no longer used in this repository and will be removed.

Please see our wiki for the new build instructions https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

Devs see CURA-9483

MariMakes, 2022-08-30:
Hey @JeanLucCoulon ,

Welcome to the Ultimaker Cura Github 🚀 
AND welcome to the Elite Building From Source Tier 👑 

@jellespijker is right, we changed the way we are building Cura and are now using Conan, but we have not cleaned up the old make scripts yet. 

You can find instructions on the new build process on: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

I'm closing this issue since it's not a bug but by design. 
Good luck and please let us know how we can help you out to build from source.

closed 2022-08-30 by MariMakes

JeanLucCoulon, 2022-08-31:
Hi,
Thanks.
I manage easily to build Cura from source (Cura/Uranium in fact) following the instructions.
I didn't know about conan. Well, it works, seems to me a bit "heavy" for only one package.

Jean-Luc

jellespijker, 2022-08-31:
Great to hear it worked with minimal effort for you.

It isn't one package tho ;-) We manage something like 12 Ultimaker owned dependencies with Conan and nearly 20 external dependencies and that is just on the C++ side. Not even counting the ~30 Python dependencies and they're all either direct or indirect dependencies of Cura. So yeah the end result is one package but using Conan actually makes this all manageable and reproducible across operating systems, CI/CD and development environments.

Let me know if you run into any issue later on or have suggestions on the workflow during development and I hope you get to write some sweet Cura features.

iggy, 2022-09-17:
Hit this trying to package Cura for a distro. The above linked directions seem very specific to developers running one-off from git. Are there instructions for distro packagers to follow?

jellespijker, 2022-09-21:
@iggy we build the installer in this GH Action. 
https://github.com/Ultimaker/Cura/blob/main/.github/workflows/cura-installer.yml

I don't know which distro you try to create a package for, But if you want to install Cura in the user space and you meet the minimum requirements. for building (compiler, cmake, python, sip, ninja, etc.)

running 
```bash
conan install cura/<version_you_want_to_install>@_/_ --build=missing --update -if cura_inst -g VirtualPythonEnv --json "cura_inst/conan_install_info.json"
source cura_inst/bin/activate
pyinstaller ./cura_inst/Ultimaker-Cura.spec
#copy the standalone App the desired location e.q.: /usr/...
``

If you want to use libs which are already installed on the system, then it becomes a different ball game and that is something that would take a lot of effort to change, something we currently don't have the resources for.