Ultimaker/Cura #13141: Bottom slider is missing

quickkurt, 2022-08-27:
### Application Version

5.1

### Platform

windows 11

### Printer

jgmaker 

### Reproduction steps

restarted cura
reinstalled

### Actual results

still missing bottom slider

### Expected results

expected the bottom slider to be there

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

bottom slider missing in 5.1.0
[image]
.

MariMakes, 2022-08-30:
Hey @quickkurt,

Welcome to the Ultimaker Cura Github 🚀 

Thanks for the report and the screenshot. 
It looks like a significant chunk is cut off the bottom of your screen including the slider. 
This is what it looks like for me, the blue box is where your window seems to be cut off. 
[image]

What kind of device are you running on? 
[Is your taskbar locked or unlocked?](https://support.microsoft.com/en-us/windows/how-to-use-the-taskbar-in-windows-0657a50f-0cc7-dbfd-ae6b-05020b195b07#:~:text=Lock%20and%20unlock%20the%20taskbar&text=on%20the%20desktop.-,Press%20and%20hold%20or%20right%2Dclick%20any%20empty%20space%20on,turn%20on%20Lock%20the%20taskbar.) 
Can you see the slider if you resize your window? 

It doesn't look like a Cura issue to me. Can you let me know if any of these work for you so we can close the issue? 😊

quickkurt, 2022-09-05:
Also I am running windows 11 if that helps.

Thanks
Kurt

Confidentiality Notice
This message and any accompanying attachments may contain confidential and/or privileged information. The message and attachments should be read and retained by intended recipients only. If you received this message in error, please notify the sender immediately and delete the message and any attachments.

[13141]<https://user-images.githubusercontent.com/40423138/187468724-b2f03d40-d447-491f-9fac-582ddc702319.PNG>

MariMakes, 2022-09-13:
Hey @quickkurt,

It's indeed helpful to know that you are running Windows 11.

Did any of the listed solutions help you? 
1. [Is your taskbar locked or unlocked?](https://support.microsoft.com/en-us/windows/how-to-use-the-taskbar-in-windows-0657a50f-0cc7-dbfd-ae6b-05020b195b07#:~:text=Lock%20and%20unlock%20the%20taskbar&text=on%20the%20desktop.-,Press%20and%20hold%20or%20right%2Dclick%20any%20empty%20space%20on,turn%20on%20Lock%20the%20taskbar.)
2. Can you see the slider if you resize your window?

Another thing that jumped to mind is. Are you in compatibility mode?
You can find this in the preferences, here:
[image]

Can you let me know if any of these solutions work for you so we can close the issue? 😊

quickkurt, 2022-09-13:
HI there , the force layer compatibility being unchecked did the trick.. it is all looking good now’’

Thanks
Kurt

Confidentiality Notice
This message and any accompanying attachments may contain confidential and/or privileged information. The message and attachments should be read and retained by intended recipients only. If you received this message in error, please notify the sender immediately and delete the message and any attachments.

[image]<https://user-images.githubusercontent.com/40423138/189872840-3d974014-1975-48a9-824f-f62b84adbfc0.png>

MariMakes, 2022-09-14:
Thanks for letting us know!
I'm happy we found what did the trick. 
I'll close the ticket.

closed 2022-09-14 by MariMakes

quickkurt, 2022-10-11:
HI there I changed refresh rate , resolution and magnification and still no bottom slider as per screen shot. This is a screen shot after my task bar drops down. I have already reinstalled 2 times and still no bottom slider????

***@***.***
Thanks
Kurt

Confidentiality Notice
This message and any accompanying attachments may contain confidential and/or privileged information. The message and attachments should be read and retained by intended recipients only. If you received this message in error, please notify the sender immediately and delete the message and any attachments.

MariMakes, 2022-10-11:
Hey @quickkurt,

Thank you for the update, but I cannot see the files that you shared. 

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.