Ultimaker/Cura #13143: sauvegarde auto

aliasze, 2022-08-28:
### Application Version

depuis 5.0

### Platform

win 11

### Printer

cr6 se

### Reproduction steps

à chaque ouverture de Cura depuis la version 5.0 la sauvegarde automatique ce met en erreur , je suis obligé de lancer la sauvegarde en manuel si je veux créer un backup

### Actual results

en manuel cela fonctionne

### Expected results

sauvegarde automatique fonctionnelle

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9439792/cura.log)

MariMakes, 2022-11-01:
Hey @aliasze,

Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I've tried on multiple occasions to recreate your error but I haven't been able to reproduce it. 
But, I've noticed this error in your logs  ERROR - [JobQueueWorker [1]] cura.Backups.BackupsManager._disableAutoSave [71]: Unable to disable the autosave as application init has not been completed

I can bring it up to the team to see if they can figure out what is going on.
Fingers Crossed 🤞

casperlamboo, 2022-12-14:
Hi @aliasze,

Thank you for the bug report. We have looked into the logs but did not find any indication what could have caused the issue. We will close the issue for now but we'll keep an eye open for similar reports.

Hope you understand our decision. Feel free to re-open the issue if you have more information and/or have reproductions steps for the issue.

Thanks,
Casper